rust-embedded/aarch32 #19: cortex-r: `_critical_section_1_0_acquire` and `_critical_section_1_0_release` are undefined

xobs, 2025-03-28:
I'm not sure why, but when I try building with `cortex-ar = {version = "0.1.0", features = ["critical-section-single-core"]}` I get errors such as "rust-lld: error: undefined symbol: _critical_section_1_0_acquire".

I can confirm that the critical section code is being built, but it's not present in the final binary.

I've managed to work around the issue by reimplementing it in my `main`, but from what I understand it should already work.

There's no LTO going on. I'm building for `armebv7r-none-eabi` with Rust 1.85.1.

Workaround:

```rust
#[no_mangle]
pub fn _critical_section_1_0_acquire() -> bool {
    use core::sync::atomic;
    // the i bit means "masked"
    let was_active = !cortex_ar::register::Cpsr::read().i();
    cortex_ar::interrupt::disable();
    atomic::compiler_fence(atomic::Ordering::SeqCst);
    was_active
}

#[no_mangle]
pub fn _critical_section_1_0_release(was_active: bool) {
    use core::sync::atomic;
    // Only re-enable interrupts if they were enabled before the critical section.
    if was_active {
        atomic::compiler_fence(atomic::Ordering::SeqCst);
        // Safety: This is OK because we're releasing a lock that was
        // entered with interrupts enabled
        unsafe {
            cortex_ar::interrupt::enable();
        }
    }
}
```

xobs, 2025-03-28:
The project I'm working on, as it stands: https://github.com/xobs/tms570-flash-algorithm/commit/5358a9a39ac4d9b97806c85eca113b004fce1769

jonathanpallant, 2025-03-28:
Huh

https://github.com/rust-embedded/cortex-ar/pull/20

jonathanpallant, 2025-03-28:
Try:

```diff
diff --git a/src/main.rs b/src/main.rs
index 8744df0..e9f367a 100644
--- a/src/main.rs
+++ b/src/main.rs
@@ -1,32 +1,9 @@
 #![no_std]
 #![no_main]

-#[no_mangle]
-pub fn _critical_section_1_0_acquire() -> bool {
-    use core::sync::atomic;
-    // the i bit means "masked"
-    let was_active = !cortex_ar::register::Cpsr::read().i();
-    cortex_ar::interrupt::disable();
-    atomic::compiler_fence(atomic::Ordering::SeqCst);
-    was_active
-}
-
-#[no_mangle]
-pub fn _critical_section_1_0_release(was_active: bool) {
-    use core::sync::atomic;
-    // Only re-enable interrupts if they were enabled before the critical section.
-    if was_active {
-        atomic::compiler_fence(atomic::Ordering::SeqCst);
-        // Safety: This is OK because we're releasing a lock that was
-        // entered with interrupts enabled
-        unsafe {
-            cortex_ar::interrupt::enable();
-        }
-    }
-}
-
 use flash_algorithm::*;
 use rtt_target::{rprintln, rtt_init_print};
+use cortex_ar as _;

 struct Algorithm;
```

closed 2025-03-28 by xobs

jonathanpallant, 2025-03-28:
yeah, it's perhaps a failure of the critical-section design. I don't know how else to work around it, especially in a tiny crate like a flash algorithm that doesn't even bring in a cortex-foo-rt crate.

adamgreig, 2025-03-28:
This is the same problem that panic handlers and crates like `defmt_rtt` have, right? If a crate provides some symbol that's only used via a non-Rust call then rustc doesn't know how to see that it's required and omits the crate. It might be something that can be improved on the Rust side, either as a detection or some crate metadata to say "include me even if I'm not obviously being called".